SciML/ModelingToolkitStandardLibrary.jl #5: Ideal op amp pin reference

ggkountouras, 2021-11-04:
In the [ideal op amp model](https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/main/src/Electrical/Analog/ideal_components.jl#L57), is the first branch the input side (i.e. between non-inverting and inverting inputs), and the second branch between the output and a reference node (typically ground)?

Usually I see op amps with 3 pins (positive input, negative input, output) or 5 pins (with positive/negative rails). This seems to be using a Voltage Controlled Voltage Source (VCVS) formulation.

ChrisRackauckas, 2021-11-04:
@ven-k?

ven-k, 2021-11-05:
It's a two port nullator-norator model i.e `v1` and `i1` can are fixed to 0 (nullator) and `v2` and `i2` can be of any value (norator).

Using the nullator simplifies the analysis. In ideal condition the opamp with neg feedback draws no current and has no voltage. And the norator ensures that even with 0 input, output can be anything.
The output voltage is the difference between `p2` and `n2` (positive and negative right ports). You can ground the `n2` for this to work like a 3-pin opamp. 

This is similar to Modelica's [IdealOpAmp](https://doc.modelica.org/Modelica%204.0.0/Resources/helpWSM/Modelica/Modelica.Electrical.Analog.Ideal.IdealOpAmp.html)

closed 2023-05-07 by ChrisRackauckas