SciML/ModelingToolkitStandardLibrary.jl #8: Make real docs

ChrisRackauckas, 2021-11-07:
This library should probably be done with extensive docstrings and summary pages.

AayushSabharwal, 2021-11-07:
I'm working on some docstrings in #7. It would probably be useful to have a CI workflow with documenter too

ValentinKaisermayer, 2022-04-23:
Nice! Apart from the magnetic stuff, which seems to not be setup properly

ChrisRackauckas, 2022-04-23:
Any follow up can be more specific. Cheers 🎉

closed 2022-04-23 by ChrisRackauckas

ChrisRackauckas, 2022-04-23:
[image]

ChrisRackauckas, 2022-04-23:
[image]

ChrisRackauckas, 2022-04-23:
[image]

ChrisRackauckas, 2022-06-04:
[image]