SciML/ModelingToolkitStandardLibrary.jl #13: Register

ChrisRackauckas, 2021-11-09:
@JuliaRegistrator register()

JuliaRegistrator, 2021-11-09:
Registration pull request created: [JuliaRegistries/General/48454](https://github.com/JuliaRegistries/General/pull/48454)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.1.0 -m "<description of version>" 3faad0e6bac1890878f985537f1dfe9fc22a5e4e
git push origin v0.1.0
```

closed 2021-11-23 by ChrisRackauckas

ChrisRackauckas, 2021-12-23:
@JuliaRegistrator register()

JuliaRegistrator, 2021-12-23:
Registration pull request created: [JuliaRegistries/General/51103](https://github.com/JuliaRegistries/General/pull/51103)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.1.1 -m "<description of version>" 2b91700050a152e1ba93b755322a7205e4d39e39
git push origin v0.1.1
```

JuliaRegistrator, 2022-04-23:
Registration pull request created: [JuliaRegistries/General/58999](https://github.com/JuliaRegistries/General/pull/58999)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.0.0 -m "<description of version>" 4c5789890f4d43c2759f8a1fcbd55c5670ebd0e9
git push origin v1.0.0
```

JuliaRegistrator, 2022-04-23:
Registration pull request created: [JuliaRegistries/General/59010](https://github.com/JuliaRegistries/General/pull/59010)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.1.0 -m "<description of version>" bdb20b5f37dda0c254b5c7e6e7d84ca02dc499f7
git push origin v1.1.0
```

JuliaRegistrator, 2022-05-21:
Registration pull request created: [JuliaRegistries/General/60711](https://github.com/JuliaRegistries/General/pull/60711)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.4.0 -m "<description of version>" 7d7901b6020ecfae79b74f067351d8cd408fd44f
git push origin v1.4.0
```
Also, note the warning: Version 1.4.0 skips over 1.2.0
This can be safely ignored. However, if you want to fix this you can do so. Call register() again after making the fix. This will update the Pull request.

JuliaRegistrator, 2022-05-21:
Registration pull request created: [JuliaRegistries/General/60713](https://github.com/JuliaRegistries/General/pull/60713)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.2.0 -m "<description of version>" 5ff99e55fe519023c57dd5d50250dd441614d5aa
git push origin v1.2.0
```

JuliaRegistrator, 2022-06-03:
Registration pull request created: [JuliaRegistries/General/61658](https://github.com/JuliaRegistries/General/pull/61658)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.3.0 -m "<description of version>" 64126f4858832528f2d8b41ceabfc7f10f86658d
git push origin v1.3.0
```

JuliaRegistrator, 2022-06-03:
Registration pull request created: [JuliaRegistries/General/61662](https://github.com/JuliaRegistries/General/pull/61662)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.3.1 -m "<description of version>" 66d11d3bb4aa6cfdd52f330a07aedc7fecc51d81
git push origin v1.3.1
```

JuliaRegistrator, 2022-06-09:
Registration pull request created: [JuliaRegistries/General/62013](https://github.com/JuliaRegistries/General/pull/62013)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.4.1 -m "<description of version>" 2f830977061ecb2562a8228829fce49492579ca4
git push origin v1.4.1
```
Also, note the warning: Version 1.4.1 skips over 1.4.0
This can be safely ignored. However, if you want to fix this you can do so. Call register() again after making the fix. This will update the Pull request.

JuliaRegistrator, 2022-06-09:
Registration pull request created: [JuliaRegistries/General/62014](https://github.com/JuliaRegistries/General/pull/62014)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.4.0 -m "<description of version>" 538fb2b1156c28c71524a8b18fdf9bd07545e39d
git push origin v1.4.0
```

JuliaRegistrator, 2022-07-30:
Registration pull request created: [JuliaRegistries/General/65335](https://github.com/JuliaRegistries/General/pull/65335)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v1.5.0 -m "<description of version>" 56eb837e38e7d196c76ae990937b233a0cce562f
git push origin v1.5.0
```

JuliaRegistrator, 2023-07-19:
Registration pull request created: [JuliaRegistries/General/87827](https://github.com/JuliaRegistries/General/pull/87827)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.0.0 -m "<description of version>" 1a81b52f71a0137919608ec495523c7e1e2f5d77
git push origin v2.0.0
```

JuliaRegistrator, 2023-07-22:
Registration pull request created: [JuliaRegistries/General/88095](https://github.com/JuliaRegistries/General/pull/88095)

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.1.0 -m "<description of version>" c7b207d531cb6bf2629a896b5f6da2b2f98bd707
git push origin v2.1.0
```

JuliaRegistrator, 2024-02-22:
Registration pull request created: [JuliaRegistries/General/101408](https://github.com/JuliaRegistries/General/pull/101408)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.4.0 -m "<description of version>" 1fbd1d0d51dd3df63f065aff395ff1ee95c1b360
git push origin v2.4.0
```

JuliaRegistrator, 2024-02-23:
Registration pull request created: [JuliaRegistries/General/101509](https://github.com/JuliaRegistries/General/pull/101509)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.5.0 -m "<description of version>" 35a41f7c2e70853527ec96b6f3a23fa882aafc87
git push origin v2.5.0
```

JuliaRegistrator, 2024-03-04:
Registration pull request created: [JuliaRegistries/General/102234](https://github.com/JuliaRegistries/General/pull/102234)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.6.0 -m "<description of version>" 918d57797c6bc90eef7101705cd7daf2d9aef537
git push origin v2.6.0
```

JuliaRegistrator, 2024-04-18:
Registration pull request created: [JuliaRegistries/General/105173](https://github.com/JuliaRegistries/General/pull/105173)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.7.0 -m "<description of version>" 7d7d324f8127a1817e1863852df578e2c8bb0ad3
git push origin v2.7.0
```

JuliaRegistrator, 2024-04-29:
Registration pull request created: [JuliaRegistries/General/105827](https://github.com/JuliaRegistries/General/pull/105827)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.7.1 -m "<description of version>" e6ffefa4d58f2e24c53f644715198871560e1080
git push origin v2.7.1
```

JuliaRegistrator, 2024-06-12:
Registration pull request created: [JuliaRegistries/General/108785](https://github.com/JuliaRegistries/General/pull/108785)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.7.2 -m "<description of version>" 22d488c2a888ae9e555c2654e379b528ce71031f
git push origin v2.7.2
```

JuliaRegistrator, 2024-07-08:
Registration pull request created: [JuliaRegistries/General/110675](https://github.com/JuliaRegistries/General/pull/110675)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.7.3 -m "<description of version>" e446d0ea76c593c1c40d24843555f5dbd178ba4c
git push origin v2.7.3
```

JuliaRegistrator, 2024-07-08:
Registration pull request created: [JuliaRegistries/General/110676](https://github.com/JuliaRegistries/General/pull/110676)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.7.4 -m "<description of version>" 27d80f5ce496165c9a89e36ba025e7496a6ab80a
git push origin v2.7.4
```

JuliaRegistrator, 2024-07-12:
Registration pull request created: [JuliaRegistries/General/110953](https://github.com/JuliaRegistries/General/pull/110953)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.7.5 -m "<description of version>" 73cef792e19cab28f3ef46dd84fe3003a49462b9
git push origin v2.7.5
```

JuliaRegistrator, 2024-07-22:
Registration pull request created: [JuliaRegistries/General/111515](https://github.com/JuliaRegistries/General/pull/111515)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.8.0 -m "<description of version>" b6c7cd273d63c4b074cd9239db901f4c3200336f
git push origin v2.8.0
```

JuliaRegistrator, 2024-07-23:
Registration pull request created: [JuliaRegistries/General/111609](https://github.com/JuliaRegistries/General/pull/111609)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.9.0 -m "<description of version>" 3d5997bcf45c380e8ee5eddc1c13b4e31ccb77f4
git push origin v2.9.0
```

JuliaRegistrator, 2024-07-26:
Registration pull request created: [JuliaRegistries/General/111822](https://github.com/JuliaRegistries/General/pull/111822)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.10.0 -m "<description of version>" 791b2274be16dfe26029c6fe5f90ad90fc37278f
git push origin v2.10.0
```

JuliaRegistrator, 2024-08-11:
Registration pull request created: [JuliaRegistries/General/112886](https://github.com/JuliaRegistries/General/pull/112886)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.11.0 -m "<description of version>" 4ef5e82b7fbea0874872f931f90f5d25a5a298fc
git push origin v2.11.0
```

JuliaRegistrator, 2024-09-12:
Registration pull request created: [JuliaRegistries/General/115033](https://github.com/JuliaRegistries/General/pull/115033)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.12.0 -m "<description of version>" f839b67374684b70e50017d734bc125fcf1d53a6
git push origin v2.12.0
```

JuliaRegistrator, 2024-09-25:
Registration pull request created: [JuliaRegistries/General/115935](https://github.com/JuliaRegistries/General/pull/115935)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.14.0 -m "<description of version>" efdff2fda2840a1de306d6087be05a7875b6b116
git push origin v2.14.0
```

JuliaRegistrator, 2024-10-04:
Registration pull request created: [JuliaRegistries/General/116588](https://github.com/JuliaRegistries/General/pull/116588)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.15.0 -m "<description of version>" 37b5157a02a2e468e943a66409d8457551e69c45
git push origin v2.15.0
```

JuliaRegistrator, 2024-10-16:
Registration pull request created: [JuliaRegistries/General/117392](https://github.com/JuliaRegistries/General/pull/117392)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.15.1 -m "<description of version>" 134df80270c9e0f52a8b6489e6d3c0b85a8762b1
git push origin v2.15.1
```

JuliaRegistrator, 2024-10-19:
Registration pull request created: [JuliaRegistries/General/117656](https://github.com/JuliaRegistries/General/pull/117656)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.16.0 -m "<description of version>" 1cef2c15dc4154d43f85ed6ce20cf1794e80a778
git push origin v2.16.0
```

JuliaRegistrator, 2024-10-31:
Registration pull request created: [JuliaRegistries/General/118460](https://github.com/JuliaRegistries/General/pull/118460)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.17.0 -m "<description of version>" 5ff27c8cb58439d2b38703af989f0befd593ae1a
git push origin v2.17.0
```

JuliaRegistrator, 2024-12-09:
Registration pull request created: [JuliaRegistries/General/121051](https://github.com/JuliaRegistries/General/pull/121051)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.18.0 -m "<description of version>" be8d22ff4eaa7e257ac4f408c34c8fdc54d14359
git push origin v2.18.0
```

JuliaRegistrator, 2025-03-21:
Registration pull request created: [JuliaRegistries/General/127392](https://github.com/JuliaRegistries/General/pull/127392)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.19.1 -m "<description of version>" 37d0d86df23291325d81cede387f0bed2e5160c9
git push origin v2.19.1
```

AayushSabharwal, 2025-05-01:
@JuliaRegistrator register()

JuliaRegistrator, 2025-05-01:
Registration pull request created: [JuliaRegistries/General/130142](https://github.com/JuliaRegistries/General/pull/130142)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.20.0 -m "<description of version>" aa55a0c6a95da5abac5fffd3b1454b3a5835c0ec
git push origin v2.20.0
```

JuliaRegistrator, 2025-05-07:
Registration pull request created: [JuliaRegistries/General/130454](https://github.com/JuliaRegistries/General/pull/130454)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.21.0 -m "<description of version>" 6244cda90bdaa639fc29ef042552b46a6e225526
git push origin v2.21.0
```

JuliaRegistrator, 2025-05-13:
Registration pull request created: [JuliaRegistries/General/130846](https://github.com/JuliaRegistries/General/pull/130846)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.21.1 -m "<description of version>" e3a049ed8857b5f79587ab2268e5fdb022870661
git push origin v2.21.1
```

JuliaRegistrator, 2025-05-29:
Registration pull request created: [JuliaRegistries/General/131943](https://github.com/JuliaRegistries/General/pull/131943)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.22.0 -m "<description of version>" 6218859ac75c32aa921a447fa82700f4457b75a0
git push origin v2.22.0
```

JuliaRegistrator, 2025-07-15:
Registration pull request created: [JuliaRegistries/General/134796](https://github.com/JuliaRegistries/General/pull/134796)

### Recommendation: add release notes

Release notes help your users know what has changed in the new version. For breaking releases in particular, it's recommended to document what breaking changes were made (or point to where changes can be found, such as a changelog).

To add release notes, just add markdown formatted text underneath the comment after the text "Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add release notes to this registration, just re-invoke JuliaRegistrator and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.25.0 -m "<description of version>" cb592ce23f3979cbd131cd090f6c08cad9e8f9c4
git push origin v2.25.0
```

JuliaRegistrator, 2025-12-24:
Registration pull request created: [JuliaRegistries/General/145002](https://github.com/JuliaRegistries/General/pull/145002)

### Recommendation: add release notes

Release notes help your users know what has changed in the new version. For breaking releases in particular, it's recommended to document what breaking changes were made (or point to where changes can be found, such as a changelog).

To add release notes, just add markdown formatted text underneath the comment after the text "Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add release notes to this registration, just re-invoke JuliaRegistrator and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v2.26.0 -m "<description of version>" 4f5e3fe4d03c75c83dbf12b600568c4c0dcd4817
git push origin v2.26.0
```