SciML/ModelingToolkitStandardLibrary.jl #70: RC circuit demonstration doesn't work

dodoplus, 2022-06-09:
The example uses `ConstantVoltage` which seems to be missing from latest release of the library.

Example is here: http://mtkstdlib.sciml.ai/dev/tutorials/rc_circuit/ as well as on the home page.

kakila, 2022-06-09:
The sources have changed, you need to create the input voltage using Blocks

```julia
 using ModelingToolkit, DifferentialEquations, Plots
 using ModelingToolkitStandardLibrary.Electrical
 using ModelingToolkitStandardLibrary.Blocks: Constant

resistance = 1.0
capacitance = 1.0
voltage = 1.0
@variables t
@named R = Resistor(R=resistance)
@named C = Capacitor(C=capacitance)
@named V = Voltage()
@named ground = Ground()
@named source = Constant(k=voltage)

rc_eqs = [
connect(V.p, R.p)
connect(R.n, C.p)
connect(C.n, V.n, ground.g)
connect(V.V, source.output)
]
@named rc_model = ODESystem(rc_eqs, t, systems=[R, C, V, source, ground])

sys = structural_simplify(rc_model)

prob = ODEProblem(sys, Pair[], (0, 10.0), [R.R => _R], jac = true)
sol = solve(prob)

p = plot(sol, vars = [C.v,R.i],
	title = "RC Circuit Demonstration",
	labels = ["Capacitor Voltage" "Resistor Current"])
```

ChrisRackauckas, 2022-06-09:
@ValentinKaisermayer can you enable doctests? That would've caught this.

ChrisRackauckas, 2022-06-09:
What do you mean?

ChrisRackauckas, 2022-06-09:
We might be fine this time, but this is the turning point. You did something breaking and two people instantly noticed it. If two people noticed it in hours, that means that the library has quite a bit of uptake by now. That's actually a bit surprising to me (I thought it was still in the "early pre-adoption phase"), but this makes it clear that it's adopted so it needs to start following the standard practices associated with a library people are have put into their dependency tree.

And on this note, I should probably prepare a release announcement 😅

ValentinKaisermayer, 2022-06-22:
Can be closed