SciML/ModelingToolkitStandardLibrary.jl #79: Propagation of variable names and variable comments

baggepinnen, 2022-06-17:
Our library components, like 
```
Blocks.StateSpace
```
currently chooses names for, e.g., the input and output variables, `u[1:3], y[1:2]` etc.
I'd like to have to option to either provide these names upon construction or change them after construction:
```
u[1:3] -> [:fan_speed, :heater_duty_cycle, :solar_irradiance]
```
In addition, one could associate comments with variables like in modelica. This would be nice irrespective of the ability to prvide variable names since comments can contain much more information than a variable name

ValentinKaisermayer, 2022-06-17:
Via the meta data system?

baggepinnen, 2022-06-17:
That was what I had in mind at least, not sure if there are better ways?