SciML/ModelingToolkitStandardLibrary.jl #98: Error in running the Chua Circuit example

RajDandekar, 2022-08-18:
I tried to run the Chua circuit example here: https://mtkstdlib.sciml.ai/stable/tutorials/custom_component/

I get an error in the last line ```structural_simplify``` of the code below
```
@parameters t

function NonlinearResistor(;name, Ga, Gb, Ve)
    @named oneport = OnePort()
    @unpack v, i = oneport
    pars = @parameters Ga=Ga Gb=Gb Ve=Ve
    eqs = [
        i ~ ifelse(v < -Ve,
                Gb*(v + Ve) - Ga*Ve,
                ifelse(v > Ve,
                    Gb*(v - Ve) + Ga*Ve,
                    Ga*v,
                ),
            )
    ]
    extend(ODESystem(eqs, t, [], pars; name=name), oneport)
end

@named L = Inductor(L=18)
@named Ro = Resistor(R=12.5e-3)
@named G = Conductor(G=0.565)
@named C1 = Capacitor(C=10, v_start=4)
@named C2 = Capacitor(C=100)
@named Nr = NonlinearResistor(
    Ga = -0.757576,
    Gb = -0.409091,
    Ve=1)
@named Gnd = Ground()

connections = [
    connect(L.p, G.p)
    connect(G.n, Nr.p)
    connect(Nr.n, Gnd.g)
    connect(C1.p, G.n)
    connect(L.n, Ro.p)
    connect(G.p, C2.p)
    connect(C1.n, Gnd.g)
    connect(C2.n, Gnd.g)
    connect(Ro.n, Gnd.g)
]

@named model = ODESystem(connections, t, systems=[L, Ro, G, C1, C2, Nr])

sys = structural_simplify(model)
```

```
ExtraVariablesSystemException: The system is unbalanced. There are 18 highest order derivative variables and 17 equations.
More variables than equations, here are the potential extra variable(s):
 L₊i(t)
 C1₊v(t)
 C2₊v(t)
 Gnd₊g₊i(t)
```

ValentinKaisermayer, 2022-08-18:
I think the problem is that `Gnd` is missing from the `systems`.

Strange that this worked on MTK v8.11.6

@YingboMa can we have a better error msg for this. Obviously there is a connection with a connector that is not in any system in `systems`.

YingboMa, 2022-09-01:
Sure. Open an issue in MTK.

closed 2022-09-01 by YingboMa