SciML/ModelingToolkitStandardLibrary.jl #101: Equivalent Circuit Model of Electrochemical/Solid state Cells

dynamic-queries, 2022-08-27:
Lithium Ion Battery Models using Equivalent Circuit Models could be a welcome addition to the features in this Standard Library.

This introductory idea from Boulder introduces the idea.
http://mocha-java.uccs.edu/ECE5710/ECE5710-Notes02.pdf 

@rkurchin Could you suggest more references and requirements for a preliminary mock-up ?

rkurchin, 2022-08-29:
I'm not really an expert in ECM's to be honest...maybe @abillscmu would have some good ideas on this front? And/or someone who does EIS...

matthew-kapp, 2025-03-16:
I wanted to look into this. To follow the rest of the library concept of inspired by MSL, I believe it makes sense to start with the existing battery model in MSL.Electrical