SciML/ModelingToolkitStandardLibrary.jl #109: Add Mechanical/Translational libraries with both *position* and *velocity* based connectors

bradcarman, 2022-09-17:
Plan:  

- Put the velocity connector approach under "Mechanical/Translational" and position connector approach under "Mechanical/TranslationalPosition" (or maybe "Mechanical/Positional"). 
- Add documentation to explain the theory and detail the deferences.  Will serve as a guide to aid in the decision of which approach to use for one's model.

Both libraries will be as similar as possible, with the only difference being the connector variables.  Because both libraries will export the same names, mixed use can/should only be done with namespaced use, although mixed use should likely be a rare case.  

A start to the documentation can be found here:  [https://github.com/bradcarman/ModelingToolkitStandardLibrary.jl/blob/main/docs/connections.md](https://github.com/bradcarman/ModelingToolkitStandardLibrary.jl/blob/main/docs/connections.md)

ValentinKaisermayer, 2022-09-17:
Can you add and demonstrate a component that measures the position for the velocity-based approach?

ValentinKaisermayer, 2022-09-17:
The rotational library follows the Modelica principles with absolute angle and torque and not the Simscape one with angular velocity and torque. Would you propose to make a separate library too in this case?

bradcarman, 2022-09-17:
Yes, I'll be explaining the position measurement in my documentation next.  And I'll plan to open a separate PR for the Mechanical/Rotational components.  For now I'll focus on the Translational part to see if everyone agrees with my idea.

PR #114 (2022-10-04): Start to Translational library with both Velocity and Position based connectors
Closes #109
closed 2022-10-04 by YingboMa