SciML/ModelingToolkitStandardLibrary.jl #123: how to pass exogenous input signals to components?

YangyangFu, 2022-10-27:
For most systems, there will be exogenous input signals as control inputs/disturbances, such as the outdoor air temperature.

Say I have a zone temperature file saved as a csv, and the temperature is sampled every hour. But I have a system, which runs at a time step of 5 min. How can I connect this exogenous temperature to `Thermal.PrescribedTemperature`, and solve the resulting ODE on a 5-minute basis? 

In Modelica, there is a block called `TimeTable`, which performs auto interpolation between time indexes. Is there any similar block in `ModelingToolkitStandardLibrary` for this purpose?

ValentinKaisermayer, 2022-10-29:
Unfortunately there is no component to do this at the moment but there should be. Or at least a tutorial to demonstrate a solution to this not uncommon task. 

You have two options:
- You can make a component where a variable is in the state vector and you change it externally. You can force this by setting `input=true`. You can use a custom callback function to update this value every sample time instance.  Maybe the code snippets in this older issue helps a little. MTK changed in the mean time but you can get the idea. https://github.com/SciML/ModelingToolkit.jl/issues/1180
- Probably easier but I am not sure if it is working in MTK would be to make a component, using a `DiscreteUpdate` operator and use a registered function that interpolates your data. Similar to this https://mtk.sciml.ai/stable/tutorials/ode_modeling/#Specifying-a-time-variable-forcing-function

YangyangFu, 2022-10-30:
Thanks a lot for this very help suggestions. I will take a look soon.

baggepinnen, 2022-10-30:
The easiest way to incorporate data like this is probably to create an interpolation function using any of the interpolators from
https://github.com/PumasAI/DataInterpolations.jl

Have a look at how some of the sources in the standard librsry are implemented, like the sine source
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/9cfa7875e8eac8a46d8997a165f851aff888c94f/src/Blocks/sources.jl#L94
Then change the equation to use the interpolators $int(t)$ instead of $/sin(t)$

ValentinKaisermayer, 2022-10-30:
If this is implemented without a discrete update operator or callback, your data has to be somewhat smooth in oder to get correct results. Although, in the library are step sources that seem to work just fine without, so the solvers are already quite good at handling this.