SciML/ModelingToolkitStandardLibrary.jl #141: Missing modelica libraries

baggepinnen, 2023-01-12:
We're still missing a few modules from the modelica standard library:
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Electrical/Batteries
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Electrical/Machines
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Electrical/Polyphase
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Electrical/PowerConverters
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Electrical/QuasiStatic
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Clocked/BooleanSignals
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Clocked/ClockSignals
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Clocked/IntegerSignals
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Clocked/RealSignals
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/Discrete.mo
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/Tables.mo
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/Logical.mo
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/MathBoolean.mo
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/MathInteger.mo
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/Noise.mo
- https://github.com/modelica/ModelicaStandardLibrary/blob/master/Modelica/Blocks/Routing.mo
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/ComplexBlocks
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Magnetic/FundamentalWave
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Magnetic/QuasiStatic
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Fluid
- https://github.com/modelica/ModelicaStandardLibrary/tree/master/Modelica/Media

ValentinKaisermayer, 2023-03-19:
I think it would be good to have at least one library demonstrating the `Flow` variable type, e.g. a `ThermalLiquid` library as opposed to an isothermal liquid (`Hydraulic`) library for stuff where the temperate can be neglected, see #78.

ValentinKaisermayer, 2023-03-19:
Other sources:
https://reference.wolfram.com/system-modeler/#Libraries
https://www.mathworks.com/help/simscape/physical-modeling.html?s_tid=CRUX_lftnav

matthew-kapp, 2025-03-04:
@baggepinnen @ValentinKaisermayer Is the goal to copy as-is? As an example I see the Modelica Battery does not contain RC pairs, so does msl need to be matched verbatim?

ChrisRackauckas, 2025-03-04:
It would be good to start there. We can always add but a starting point of just copying is always good