SciML/ModelingToolkitStandardLibrary.jl #148: Let default constructor initialize the GUIMetadata type

zsoerenm, 2023-02-10:
Something like this would be helpful e.g.

```julia
function Resistor(; name, R)
    @named oneport = OnePort()
    @unpack v, i = oneport
    pars = @parameters R = R
    eqs = [
        v ~ i * R,
    ]
    extend(ODESystem(eqs, t, [], pars; name = name, gui_metadata = GUIMetadata("Resistor")), oneport)
end
```
Maybe a symbol would fit better than a string?

ChrisRackauckas, 2023-02-11:
This seems misguided. Will this just be the same as the type name that would then be used for better lowering. @YingboMa should we just add that first?

YingboMa, 2023-02-11:
Should we always instantiate a component like
```julia
@component function Resistor(; name, R)
    @named oneport = OnePort()
    @unpack v, i = oneport
    pars = @parameters R = R
    eqs = [
        v ~ i * R,
    ]
    extend(ODESystem(eqs, t, [], pars; name = name), oneport)
end
```
?

ChrisRackauckas, 2023-02-11:
I think so. And it would make a `Val(:Resistor)` type information or something?

YingboMa, 2023-02-11:
We then have to assume all the components are instantiated in the same world age, but I guess it's a fair assumption.

> And it would make a Val(:Resistor) type information or something?

Since it would be a very simple type system within MTK, we would use some other encoding that's independent from Julia's so that we don't take unnecessary compile time hit.

ChrisRackauckas, 2023-02-11:
oh yes no val. But yeah, let's go this route.