SciML/ModelingToolkitStandardLibrary.jl #171: ThermalConductor units issue

kakila, 2023-05-12:
The documentation specifies that G is given in units of Power per Temperature, however this line:
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/f7caceb098cfcc76ef556472fa2f3c6cc924812f/src/Thermal/HeatTransfer/ideal_components.jl#L120

Multiplies G with a factro tha tis in Power units, giving a result in Power^2 per Temperature.
I guess this line needs to be a division: Q_flow / G; or G needs to be defined differently.

kakila, 2023-05-12:
Why not define a single device that sets
```
0 ~ C * dT - Q_flow
```
and then make the *Conductor *Resistors just syntax sugar built on top of this element?
This would be less code to maintain

ven-k, 2023-05-29:
Yes `ConvectiveConductor` equation should be `Q_flow ~ G * dT`. Thanks for catching.
(Issue title needs to be fixed ThermalConductor -> (linked) ConvectiveConductor)

> Why not define a single device that sets

We can define partial blocks and extend them when `systems`, `@parameters`, `@variables` can be reused. As R and G parameters are defined using the given values of R and G, we can't do this.

closed 2023-05-30 by YingboMa

kakila, 2023-05-31:
> Yes `ConvectiveConductor` equation should be `Q_flow ~ G * dT`. Thanks for catching. (Issue title needs to be fixed ThermalConductor -> (linked) ConvectiveConductor)
>
>
> We can define partial blocks [...]

Aren't G and R reciprocals from each other?

PR #179 (2023-05-30): Fix the ConvectiveConductor equation and adds `ConvectiveElement1D`
- Closes #171 
- Adds `ConvectiveElement1D`, a partial model that defines temperature drop and heat flow rate from `solid` to `fluid`
- While here, fixes couple of typos Hydrualic -> Hydraulic