SciML/ModelingToolkitStandardLibrary.jl #206: Add DcMotor model

jevontech, 2023-08-10:
I am planning to add a DcMotor component composed of existing components to the library.
What would be the most appropriate location?
I was thinking of adding a new toplevel module ElectroMechanical (with submodules for Rotational and Translational?), but for example the EMF model is now located in Electrical. 
Any suggestions ?

closed 2023-08-11 by jevontech

ChrisRackauckas, 2023-08-11:
Hey, is this going to be a more sophisticated DC motor model or is it going to be similar to one like the Modelica standard library?

jevontech, 2023-08-13:
I want to start with the basic DC motor model as used in the ModelingToolkitStandardLibrary example and then build other models later on. For now I will put them in my own library elan8/Elan8ComponentLibrary.jl