SciML/ModelingToolkitStandardLibrary.jl #207: Add units to component parameters

bauglir, 2023-08-11:
Many components have nice descriptions for parameters and include the corresponding unit information in their comments, e.g. the [`Resistor`](https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/59446722f3ab6114e1ce0562d1e469bb3787c361/src/Electrical/Analog/ideal_components.jl#L20-L46) or the [`ConstantPermeance`](https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/59446722f3ab6114e1ce0562d1e469bb3787c361/src/Magnetic/FluxTubes/basic.jl#L68-L76). However, even though this appears to be supported by the `@mtkmodel` macro, the `unit` of these parameters is not defined. It would be nice if this additional metadata was made available within the component definitions for introspection.

I'm not sure if the same would be possible for states, but if it is then I imagine that'd be useful as well.

bauglir, 2023-11-09:
Seems related to #90.

ven-k, 2024-01-24:
#214 is switching to use DynamicQuantities. We need ability to define custom units (like Wb). The foll. PR adds that:
- https://github.com/SymbolicML/DynamicQuantities.jl/pull/107

ven-k, 2024-05-03:
Related:
- https://github.com/JuliaSymbolics/Symbolics.jl/pull/1104
- https://github.com/SciML/ModelingToolkit.jl/pull/2583