SciML/ModelingToolkitStandardLibrary.jl #237: Icons todo

david-gottschalk, 2023-10-25:
Heya,
I was going after the `export`ed `@mtkmodel`s and `@component`s

The digital gates seem to not get included https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/0f0fae138fe202140a4862eccaaac734051dc44f/src/Electrical/Electrical.jl#L29-L30
nonetheless an adjusted version is present.

Date: 25th of October 2023

# Current status [MTKSL/src](https://github.com/SciML/ModelingToolkitStandardLibrary.jl/tree/main/src)
- [ ] Blocks
  - [X] RealInput
  - [X] RealOutput
  - [X] SISO
  - [X] Gain
  - [X] Sum
  - [X] MatrixGain
  - [X] Feedback
  - [X] Add
  - [X] Add3
  - [X] Product
  - [X] Division
  - [X] Abs
  - [X] Sign
  - [X] Sqrt
  - [X] Sin
  - [X] Cos
  - [X] Tan
  - [X] Asin
  - [X] Acos
  - [X] Atan
  - [X] Atan2
  - [X] Sinh
  - [X] Cosh
  - [X] Tanh
  - [X] Exp
  - [X] Log
  - [X] Log10
  - [X] Constant
  - [ ] TimeVaryingFunction
  - [X] Sine
  - [X] Cosine
  - [ ] ContinuousClock
  - [X] Ramp
  - [X] Step
  - [X] ExpSine
  - [X] Square
  - [ ] Triangular
  - [ ] Parameter
  - [ ] SampledData ?
  - [X] Limiter
  - [X] DeadZone
  - [X] SlewRateLimiter
  - [X] Integrator
  - [X] Derivative
  - [X] FirstOrder
  - [X] SecondOrder
  - [X] StateSpace
  - [X] PI
  - [ ] LimPI
  - [X] PID
  - [X] LimPID
- [ ] Electrical
  - [X] Pin
  - [X] OnePort
  - [X] Capacitor
  - [X] Ground
  - [X] Inductor
  - [X] Resistor
  - [X] Conductor
  - [X] Short
  - [X] IdealOpAmp
  - [ ] EMF
  - [X] HeatingResistor
  - [X] CurrentSensor
  - [X] PotentialSensor
  - [X] VoltageSensor
  - [X] PowerSensor
  - [X] MultiSensor
  - [X] Voltage
  - [X] Current
- [ ] Hydraulic
  - [ ] IsothermalCompressible
    - [ ] HydraulicPort
    - [ ] HydraulicFluid
    - [ ] Cap
    - [ ] Tube
    - [ ] FixedVolume
    - [ ] DynamicVolume
    - [ ] MassFlow
    - [ ] Pressure
    - [ ] FixedPressure
- [ ] Magnetic (@annelle-abatoni)
  - [ ] FluxTubes
    - [ ] PositiveMagneticPort
    - [ ] NegativeMagneticPort
    - [ ] TwoPort
    - [ ] Ground
    - [ ] Idle
    - [ ] Short
    - [ ] Crossing
    - [ ] ConstantPermeance
    - [ ] ConstantReluctance
    - [ ] EddyCurrent
    - [ ] ElectroMagneticConverter
    - [ ] ConstantMagneticPotentialDifference
    - [ ] ConstantMagneticFlux
- [ ] Mechanical
  - [ ] MultiBody2D
    - [ ] Link
  - [ ] Rotational
    - [X] Flange
    - [X] Support
    - [X] Fixed
    - [X] Inertia
    - [X] Spring
    - [X] Damper
    - [X] SpringDamper
    - [X] IdealGear
    - [ ] RotationalFriction
    - [X] Torque
    - [ ] ConstantTorque
    - [X] Speed
    - [X] AngleSensor
    - [X] SpeedSensor
    - [X] TorqueSensor
    - [X] RelSpeedSensor
  - [X] Translational
    - [X] MechanicalPort
    - [X] Mass
    - [X] Spring
    - [X] Damper
    - [X] Fixed
    - [X] Force
    - [X] Position
    - [X] Velocity
    - [X] Acceleration
    - [X] ForceSensor
    - [X] PositionSensor
  - [X] TranslationalModelica / TranslationalPosition
    - [X] Flange
    - [X] Fixed
    - [X] Mass
    - [X] Spring
    - [X] Damper
    - [X] Force
- [ ] Thermal
  - [X] HeatPort
  - [X] Element1D
  - [ ] HeatTransfer
    - [X] BodyRadiation
    - [ ] ConvectiveConductor
    - [X] ConvectiveResistor
    - [X] HeatCapacitor,
    - [X] ThermalConductor,
    - [X] ThermalResistor
    - [X] ThermalCollector
    - [X] RelativeTemperatureSensor
    - [X] HeatFlowSensor
    - [X] TemperatureSensor
    - [X] FixedHeatFlow
    - [X] FixedTemperature
    - [X] PrescribedHeatFlow
    - [X] PrescribedTemperature

# TODO (i.e., unchecked in the list above)
- [ ] Blocks
  - [ ] TimeVaryingFunction
  - [ ] ContinuousClock
  - [ ] Triangular
  - [ ] Parameter
  - [ ] SampledData ?
  - [ ] LimPI
- [ ] Electrical
  - [ ] EMF
- [ ] Hydraulic
- [ ] Mechanical
  - [ ] MultiBody2D
    - [ ] Link
  - [ ] Rotational
    - [ ] RotationalFriction
    - [ ] ConstantTorque
- [ ] Thermal
  - [ ] HeatTransfer
    - [ ] ConvectiveConductor

These are mostly components for which I did'nt find a suitable analog in Modelica

Thanks for everything and take care,
Cheers!

matthew-kapp, 2025-03-16:
@david-gottschalk Where are these icons stored? Are they included in the model using @icon or how? And how does one view the model with icons? I don't believe there is yet like an OM Connection Editor right?

david-gottschalk, 2025-06-26:
Hey, @matthew-kapp, sorry for the super late response, haven't checked my Github in ages... my work on this was after my internship at JuliaHub and this was my goodbye message letting the team know which icons had been ported. The icons and display logic are part of [JuliaSim](https://juliahub.com/products/juliasim), i.e., not open source I believe. Due to this being so long ago I don't remember my reasoning to posting this issue here instead of in the private repo, my apologies for the confusion. I'll close this issue.
Have a nice time! 
Cheers