SciML/ModelingToolkitStandardLibrary.jl #267: Running Models with Discrete Data documentation throws errors

aditya-sengupta, 2024-02-21:
**Describe the bug 🐞**

The Running Models with Discrete Data
 [documentation page](https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/main/docs/src/tutorials/input_component.md) throws errors, and I'm not able to modify it to get to the correct output.

**Expected behavior**

The code should run and should insert ODE solutions for the problem with `data1` and `data2` in the variables `sol1` and `sol2` respectively. I tried to fix any bugs I could with minor modifications, but wasn't able to get past a certain point in trying to get this working. I passed in `name=name` to the `TimeVaryingFunction` and `SampledData` constructors, which allowed everything up to the `SampledData` section's first `get_prob` call to run. I also changed `defs[s.src.buffer]` to `defs[s.systems[1].buffer]` to get around the error `ERROR: ArgumentError: System system: variable src does not exist`. Finally, I removed the `@sync` and `@async` at the end to get a clearer error message. I wasn't able to do anything further to address the error: `no method matching getindex(::Parameter{Float64}, ::Int64)`.

**Minimal Reproducible Example 👇**

```julia
using ModelingToolkit
using ModelingToolkitStandardLibrary.Blocks
using DataInterpolations
using OrdinaryDiffEq
using Test

@parameters t
D = Differential(t)

function System(f; name)
    src = TimeVaryingFunction(f; name=name)

    vars = @variables f(t)=0 x(t)=0 dx(t)=0 ddx(t)=0
    pars = @parameters m=10 k=1000 d=1

    eqs = [f ~ src.output.u
        ddx * 10 ~ k * x + d * dx + f
        D(x) ~ dx
        D(dx) ~ ddx]

    ODESystem(eqs, t, vars, pars; systems = [src], name)
end

dt = 4e-4
time = 0:dt:0.1
data = sin.(2 * pi * time * 100) # example data

f = LinearInterpolation(data, time)

@named system = System(f)
sys = structural_simplify(system)
prob = ODEProblem(sys, [], (0, time[end]))
sol = solve(prob, ImplicitEuler())

const rdata = Ref{Vector{Float64}}()

# Data Sets
data1 = sin.(2 * pi * time * 100)
data2 = cos.(2 * pi * time * 50)

function get_sampled_data(t)
    i = floor(Int, t / dt) + 1
    x = rdata[][i]

    return x
end

Symbolics.@register_symbolic get_sampled_data(t)

function System(; name)
    vars = @variables f(t)=0 x(t)=0 dx(t)=0 ddx(t)=0
    pars = @parameters m=10 k=1000 d=1

    eqs = [f ~ get_sampled_data(t)
        ddx * 10 ~ k * x + d * dx + f
        D(x) ~ dx
        D(dx) ~ ddx]

[... 335 lines cut ...]
  [8e850b90] libblastrampoline_jll v5.8.0+1
  [8e850ede] nghttp2_jll v1.52.0+1
  [3f19e933] p7zip_jll v17.4.0+2
Info Packages marked with ⌅ have new versions available but compatibility constraints restrict them from upgrading. To see why use `status --outdated -m`
```

  - Output of `versioninfo()`

```julia
Julia Version 1.10.1
Commit 7790d6f0641 (2024-02-13 20:41 UTC)
Build Info:
  Official https://julialang.org/ release
Platform Info:
  OS: macOS (arm64-apple-darwin22.4.0)
  CPU: 10 × Apple M1 Pro
  WORD_SIZE: 64
  LIBM: libopenlibm
  LLVM: libLLVM-15.0.7 (ORCJIT, apple-m1)
Threads: 8 default, 0 interactive, 4 GC (on 8 virtual cores)
Environment:
  JULIA_NUM_THREADS = 8
  LD_LIBRARY_PATH = :/Users/adityasengupta/scetre/DAOROOT/lib:/Users/adityasengupta/scetre/DAOROOT/lib64:/opt/homebrew/lib
  JULIA_EDITOR = code
```

aditya-sengupta, 2024-02-21:
Undoing some of my changes in accordance with https://github.com/SciML/ModelingToolkitStandardLibrary.jl/issues/253 so I can just use `@named` instead of passing around a kwarg and get closer to the expected behaviour. The substantial issue remains.

aditya-sengupta, 2024-02-21:
Found a workaround, but I don't know if this is idiomatic for MTK; it uses some awkward type mixing.

```
@named system = System()
sys = structural_simplify(system)
prob = ODEProblem(sys, [], (0, time[end]); tofloat = false)

function get_prob(data)
    remake(prob; p=[prob.p[1], [Parameter(data, dt)]])
end

prob1 = get_prob(data1)
prob2 = get_prob(data2)

sol1 = Ref{ODESolution}()
sol2 = Ref{ODESolution}()

sol1[] = solve(prob1, ImplicitEuler())
sol2[] = solve(prob2, ImplicitEuler())
```

langestefan, 2024-05-20:
Hi @aditya-sengupta unfortunately the examples are still broken. Have you found a way to get the `TimeVaryingFunction Component` sample (the top one) to work? I fixed the issue with the name (`@named src = TimeVaryingFunction(f)`) but the example still doesn't run, throwing:

>MethodError: objects of type `Vector{Float64}` are not callable
Use square brackets [] for indexing an `Array`.

aditya-sengupta, 2024-05-20:
I ended up writing my use case directly in OrdinaryDiffEq - I think there
are workarounds but as far as I remember they were too complicated

langestefan, 2024-05-20:
I managed to get it running by upgrading datainterpolations.jl

ChrisRackauckas, 2024-05-20:
Yes DataInterpolations v5 should have fixed this, and I can see from this issue that it was opened with v4

closed 2024-05-20 by ChrisRackauckas