SciML/ModelingToolkitStandardLibrary.jl #282: Several components specify defaults instead of guesses

avinashresearch1, 2024-04-11:
**Describe the bug 🐞**
Several components e.g., 
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/02a134cd430b035e43c8c43596d32ecbff1a08d5/src/Electrical/utils.jl#L35
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/02a134cd430b035e43c8c43596d32ecbff1a08d5/src/Electrical/utils.jl#L72
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/02a134cd430b035e43c8c43596d32ecbff1a08d5/src/Electrical/Analog/sensors.jl#L152
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/02a134cd430b035e43c8c43596d32ecbff1a08d5/src/Blocks/continuous.jl#L548
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/02a134cd430b035e43c8c43596d32ecbff1a08d5/src/Blocks/continuous.jl#L621

specify variables with defaults instead of guesses. Specifying a default in MTK v9 would mean that an explicit _constrant_ is added to the initialization problem which is often infeasible or makes the problem extremely overdetermined. This would be the equivalent of setting `fix = true and specifying a `start` value in Modelica.

avinashresearch1, 2024-04-11:
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/pull/281 fixes for _some_ Thermal and Blocks components.

avinashresearch1, 2024-05-07:
Also: should the `x` passes into `FirstOrder` be a guess not a default:

matthew-kapp, 2025-03-07:
@avinashresearch1 do the QA aqua tests accept [guess =, description =] ?

BambOoxX, 2025-03-14:
Same for Mechanical components it seems...