SciML/ModelingToolkitStandardLibrary.jl #285: Add a `Random` Block ?

BambOoxX, 2024-04-18:
I was looking at the code in https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/main/src/Blocks/sources.jl and I was wondering if you were considering adding a random source générators, e.g. white noise, white bandlimited noise, red or else. 
Given that there are also elements for modelling random processes in https://docs.sciml.ai/DiffEqNoiseProcess/stable/ I was wondering if such an addition was relevant or not. 

In my case, I'm trying to simulate a system having a controlled random input, so it is not noise, but still a random signal. 

What is your opinion on this ?

ChrisRackauckas, 2024-04-18:
Sure, just use the brownian syntax and it should be a good way to make SDEs

BambOoxX, 2024-04-18:
Ah, I think I was not completely clear, my signal follows a random sequence in time, but that sequence is defined by the random number generator.

So in a sense it's still a conventional ode but at each timestep the signal is given by `rand(Float64)`.

That does not look like noises processes, does it?

ChrisRackauckas, 2024-04-19:
It does. That's exactly what an SDE is doing, with `randn` though because if it's not mean zero then the solution technically goes to infinity infinitely fast, so I presume your case is also `randn`.

BambOoxX, 2024-04-19:
For my case, the signal is generally a band-limited gaussian white noise, of controlled amplitude, so `randn` would indeed be the sequence generator. I will look deeper into NoiceProcesses and re-open if I feel something should be directly included here. Thanks for the help !

closed 2024-04-19 by BambOoxX