SciML/ModelingToolkitStandardLibrary.jl #287: MTK StandardLibrary v2.7.0 broke SecondOrder-block initialization

jaakkor2, 2024-04-19:
**Describe the example**

ODE initialization broke for `SecondOrder` block in ModelingToolkitStandardLibrary v2.7.0.  Similar usage of `FirstOrder`-block still works.

**Minimal Reproducible Example 👇**

```julia
using ModelingToolkit: t_nounits as t, ODESystem, connect, structural_simplify, @named
using OrdinaryDiffEq: ODEProblem
using ModelingToolkitStandardLibrary.Blocks: Sine, FirstOrder, SecondOrder

@named source = Sine(frequency = 1)
@named fo = FirstOrder(T = 1)
@named so = SecondOrder()
    
u0 = [
    ]

connections = [
    connect(source.output, fo.input)
    connect(source.output, so.input)
]

@named sys = ODESystem(connections, t, systems = [source, fo, so])

sys = structural_simplify(sys)
tspan = (0.0, 5.0)
par = Pair[]
prob = ODEProblem(sys, u0, tspan, par)
```

**Error & Stacktrace ⚠️**

```julia
ERROR: Initialization incomplete. Not all of the state variables of the
DAE system can be determined by the initialization. Missing
variables:

Any[so₊xd(t)]

Stacktrace:
  [1] ModelingToolkit.InitializationProblem{…}(sys::ODESystem, t::Float64, u0map::Dict{…}, parammap::Dict{…}; guesses::Dict{…}, check_length::Bool, warn_initialize_determined::Bool, kwargs::@Kwargs{})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1617
  [2] (ModelingToolkit.InitializationProblem{…})(::ODESystem, ::Float64, ::Vararg{…}; kwargs::@Kwargs{…})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1571
  [3] InitializationProblem
    @ C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1570 [inlined]
  [4] #InitializationProblem#790
    @ C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1559 [inlined]
  [5] InitializationProblem
    @ C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1558 [inlined]
  [6] process_DEProblem(constructor::Type, sys::ODESystem, u0map::Vector{…}, parammap::Vector{…}; implicit_dae::Bool, du0map::Nothing, version::Nothing, tgrad::Bool, jac::Bool, checkbounds::Bool, sparse::Bool, simplify::Bool, linenumbers::Bool, parallel::Symbolics.SerialForm, eval_expression::Bool, use_union::Bool, tofloat::Bool, symbolic_u0::Bool, u0_constructor::typeof(identity), guesses::Dict{…}, t::Float64, warn_initialize_determined::Bool, build_initializeprob::Bool, kwargs::@Kwargs{…})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:920
  [7] process_DEProblem
    @ C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:833 [inlined]
  [8] (ODEProblem{…})(sys::ODESystem, u0map::Vector{…}, tspan::Tuple{…}, parammap::Vector{…}; callback::Nothing, check_length::Bool, warn_initialize_determined::Bool, kwargs::@Kwargs{})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1084
  [9] (ODEProblem{…})(sys::ODESystem, u0map::Vector{…}, tspan::Tuple{…}, parammap::Vector{…})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1074
 [10] (ODEProblem{true})(::ODESystem, ::Vector{Any}, ::Vararg{Any}; kwargs::@Kwargs{})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1061
 [11] (ODEProblem{true})(::ODESystem, ::Vector{Any}, ::Vararg{Any})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1060
 [12] ODEProblem(::ODESystem, ::Vector{Any}, ::Vararg{Any}; kwargs::@Kwargs{})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1050
 [13] ODEProblem(::ODESystem, ::Vector{Any}, ::Vararg{Any})
    @ ModelingToolkit C:\Users\jaakkor2\.julia\packages\ModelingToolkit\pmsNn\src\systems\diffeqs\abstractodesystem.jl:1049
 [14] top-level scope
    @ REPL[14]:1
Some type information was truncated. Use `show(err)` to see complete types.
[... 486 lines cut ...]

`SecondOrder` is initialized with `guess` https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/v2.7.0/src/Blocks/continuous.jl#L170-L173
```
  @variables begin
        x(t), [description = "State of SecondOrder filter", guess = 0.0]
        xd(t), [description = "Derivative state of SecondOrder filter", guess = 0.0]
    end
```
whereas `FirstOrder` is not https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/v2.7.0/src/Blocks/continuous.jl#L125-L126
```
    @variables begin
        x(t) = 0.0, [description = "State of FirstOrder filter"]
    end
```

Seems to work with either
```
@named so = SecondOrder(xd = 0)
```
or
```
u0 = [
    so.xd => 0.0
    ]
```

baggepinnen, 2024-04-19:
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/pull/274

This PR was breaking but was released as non-breaking

jaakkor2, 2024-05-14:
https://docs.sciml.ai/ModelingToolkitStandardLibrary/stable/ docs still point to v.2.6.0.

Building docs is broken, here `@named C = Gain(-1)`
```
┌ Error: failed to run `@example` block in src\API\linear_analysis.md:93-96
│ ```@example LINEAR_ANALYSIS
│ matrices_L = get_looptransfer(sys, :plant_output)[1]
│ L = ss(matrices_L...)
│ ```
│   exception =
│    Initial condition underdefined. Some are missing from the variable map.
│    Please provide a default (`u0`), initialization equation, or guess
│    for the following variables:
│
│    Any[C₊input₊u(t)]
│
│    Stacktrace:
│      [1] _varmap_to_vars(varmap::Dict{Any, Any}, varlist::Vector{SymbolicUtils.BasicSymbolic{Real}}; defaults::Dict{Any, Any}, check::Bool, toterm::Function, initialization_phase::Bool)
│        @ ModelingToolkit C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkit\w72sG\src\variables.jl:205
│      [2] varmap_to_vars(varmap::SciMLBase.NullParameters, varlist::Vector{SymbolicUtils.BasicSymbolic{Real}}; defaults::Dict{Any, Any}, check::Bool, toterm::Function, promotetoconcrete::Nothing, tofloat::Bool, use_union::Bool)
│        @ ModelingToolkit C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkit\w72sG\src\variables.jl:155
│      [3] get_u0_p(sys::ODESystem, u0map::Dict{Any, Any}, parammap::SciMLBase.NullParameters; use_union::Bool, tofloat::Bool, symbolic_u0::Bool)
│        @ ModelingToolkit C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkit\w72sG\src\systems\diffeqs\abstractodesystem.jl:811
│      [4] linearization_function(sys::ODESystem, inputs::Vector{Num}, outputs::Vector{Num}; simplify::Bool, initialize::Bool, op::Dict{Any, Any}, p::SciMLBase.NullParameters, zero_dummy_der::Bool, kwargs::@Kwargs{})
│        @ ModelingToolkit C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkit\w72sG\src\systems\abstractsystem.jl:1760
│      [5] get_looptransfer_function(sys::ODESystem, ap_name::Symbol; loop_openings::Nothing, kwargs::@Kwargs{op::Dict{Any, Any}, p::SciMLBase.NullParameters})
│        @ ModelingToolkitStandardLibrary.Blocks C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkitStandardLibrary\mEg8U\src\Blocks\analysis_points.jl:348
│      [6] get_looptransfer_function
│        @ C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkitStandardLibrary\mEg8U\src\Blocks\analysis_points.jl:325 [inlined]
│      [7] get_looptransfer(::ODESystem, ::Symbol; loop_openings::Nothing, op::Dict{Any, Any}, p::SciMLBase.NullParameters, kwargs::@Kwargs{})
│        @ ModelingToolkitStandardLibrary.Blocks C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkitStandardLibrary\mEg8U\src\Blocks\analysis_points.jl:470
│      [8] get_looptransfer(::ODESystem, ::Symbol)
│        @ ModelingToolkitStandardLibrary.Blocks C:\Users\jaakkor2\MyTemp\depot1\packages\ModelingToolkitStandardLibrary\mEg8U\src\Blocks\analysis_points.jl:463
│      [9] top-level scope
│        @ linear_analysis.md:94
│     [10] eval
│        @ .\boot.jl:385 [inlined]
│     [11] #58
│        @ C:\Users\jaakkor2\MyTemp\depot1\packages\Documenter\CJeWX\src\expander_pipeline.jl:754 [inlined]
│     [12] cd(f::Documenter.var"#58#60"{Module, Expr}, dir::String)
│        @ Base.Filesystem .\file.jl:101
│     [13] (::Documenter.var"#57#59"{Documenter.Page, Module, Expr})()
│        @ Documenter C:\Users\jaakkor2\MyTemp\depot1\packages\Documenter\CJeWX\src\expander_pipeline.jl:753
│     [14] (::IOCapture.var"#4#7"{DataType, Documenter.var"#57#59"{Documenter.Page, Module, Expr}, IOContext{Base.PipeEndpoint}, IOContext{Base.PipeEndpoint}, Base.TTY, Base.TTY})()
│        @ IOCapture C:\Users\jaakkor2\MyTemp\depot1\packages\IOCapture\Rzdxd\src\IOCapture.jl:161
│     [15] with_logstate(f::Function, logstate::Any)
│        @ Base.CoreLogging .\logging.jl:515
│     [16] with_logger
│        @ .\logging.jl:627 [inlined]
│     [17] capture(f::Documenter.var"#57#59"{Documenter.Page, Module, Expr}; rethrow::Type, color::Bool, passthrough::Bool, capture_buffer::IOBuffer)
│        @ IOCapture C:\Users\jaakkor2\MyTemp\depot1\packages\IOCapture\Rzdxd\src\IOCapture.jl:158
│     [18] runner(::Type{Documenter.Expanders.ExampleBlocks}, node::MarkdownAST.Node{Nothing}, page::Documenter.Page, doc::Documenter.Document)
│        @ Documenter C:\Users\jaakkor2\MyTemp\depot1\packages\Documenter\CJeWX\src\expander_pipeline.jl:752
└ @ Documenter C:\Users\jaakkor2\MyTemp\depot1\packages\Documenter\CJeWX\src\utilities\utilities.jl:44
```

ChrisRackauckas, 2024-07-26:
The docs build is fixed. Yes, we reduced the number of hardcoded zero initializations a bunch. This one probably should have gotten a major, but it's far back enough now that reversing all of the other updates since then would be majorly breaking as well. So I believe this is completed, please feel free to continue the conversation if it's not.

closed 2024-07-26 by ChrisRackauckas