SciML/ModelingToolkitStandardLibrary.jl #328: Initialization issue in CI

baggepinnen, 2024-09-12:
CI has never been made to pass for this repo on MTKv9, an error 

> Initialization expression model₊inertia₊phi(t) is currently not supported. If its a higher order derivative expression, then only the dummy derivative expressions are supported.

appears. Introduced in #271

https://github.com/SciML/ModelingToolkitStandardLibrary.jl/actions/runs/10812682331/job/29994875171#step:6:193

ChrisRackauckas, 2024-09-12:
> CI has never been made to pass for this repo on MTKv9, an error

Here's a merge of the doc build working on July 29th:

https://github.com/SciML/ModelingToolkitStandardLibrary.jl/actions/runs/10143438690/job/28044805481#step:5:329

Saying that doc builds are failing because of a failure in February (https://github.com/SciML/ModelingToolkitStandardLibrary.jl/pull/271) and "CI has never been made to pass for this repo on MTKv9" is very easy to verify that it's not true... so let's get back to reality.

Looks like it was introduced in https://github.com/SciML/ModelingToolkitStandardLibrary.jl/pull/313 https://github.com/SciML/ModelingToolkitStandardLibrary.jl/commit/70f767cf53c10080305fac60d91b36743b07cc0d 

The doc build did pass, but it seems to be because the initialization hit the ODE case so the typo on the given initial values was ignored. https://github.com/SciML/ModelingToolkit.jl/pull/3032 exposes that typo in the tutorial, so https://github.com/SciML/ModelingToolkitStandardLibrary.jl/pull/329 looks to be a fix.

baggepinnen, 2024-09-12:
Ok, the docs build have passed, but I still think CI hasn't fully passed since MTK v9 merge, here's the test runs filtered on "success"
https://github.com/SciML/ModelingToolkitStandardLibrary.jl/actions/workflows/Downgrade.yml?query=is%3Asuccess
the last one appears to be before the v9 compat merge

ChrisRackauckas, 2024-09-12:
Downgrade can be ignored because that's a Pkg.jl issue. That's irrelevant to the CI failure.

closed 2024-09-12 by ChrisRackauckas