SciML/ModelingToolkitStandardLibrary.jl #340: Interpolations component cannot be used within @mtkmodel macro

rbborges, 2024-11-08:
**Describe the bug 🐞**

The newly created Interpolations component, available in ModelingToolkitStandardLibrary v 2.17.0 under the Blocks module, gives problems when used within the @mtkmodel macro.

**Expected behavior**

The Interpolation component should work fine when used ina model defined using the @mtkmodel macro.

**Minimal Reproducible Example 👇**

The code below is from  and example available in this [link](https://github.com/SciML/ModelingToolkitStandardLibrary.jl/blob/bca6e280efcfad0ea6187a0b2e4a861d8ff10ed0/docs/src/tutorials/input_component.md?plain=1), which works fine with Interpolation being used as a component:

```julia
using ModelingToolkit
using ModelingToolkit: t_nounits as t, D_nounits as D
using ModelingToolkitStandardLibrary
using ModelingToolkitStandardLibrary.Blocks
using DataInterpolations
using OrdinaryDiffEq
using DataFrames
using Plots

function MassSpringDamper(; name)
    @named input = RealInput()
    @variables f(t)=0 x(t)=0 dx(t)=0 ddx(t)=0
    @parameters m=10 k=1000 d=1

    eqs = [
           f ~ input.u
           ddx * 10 ~ k * x + d * dx + f
           D(x) ~ dx
           D(dx) ~ ddx]

    ODESystem(eqs, t; name, systems = [input])
end

function MassSpringDamperSystem(data, time; name)
    @named src = ModelingToolkitStandardLibrary.Blocks.Interpolation(LinearInterpolation, data, time)
    @named clk = ContinuousClock()
    @named model = MassSpringDamper()

    eqs = [
        connect(src.input, clk.output)
        connect(src.output, model.input)
    ]

    ODESystem(eqs, t, [], []; name, systems = [src, clk, model])
end

function generate_data()
    dt = 4e-4
    time = 0:dt:0.1
    data = sin.(2 * pi * time * 100)

    return DataFrame(; time, data)
end

df = generate_data() # example data

@named system = MassSpringDamperSystem(df.data, df.time)
sys = structural_simplify(system)
prob = ODEProblem(sys, [], (0, df.time[end]))
sol = solve(prob)
plot(sol)
```

Below is a toy example which does NOT work when Interpolation is used within @mtkmodel macro:

The model definition works fine:

[... 498 lines cut ...]
  [bea87d4a] SuiteSparse_jll v7.7.0+0
  [83775a58] Zlib_jll v1.2.13+1
  [8e850b90] libblastrampoline_jll v5.11.0+0
  [8e850ede] nghttp2_jll v1.59.0+0
  [3f19e933] p7zip_jll v17.4.0+2
```

  - Output of `versioninfo()`

```julia
Julia Version 1.11.1
Commit 8f5b7ca12ad (2024-10-16 10:53 UTC)
Build Info:
  Official https://julialang.org/ release
Platform Info:
  OS: Linux (x86_64-linux-gnu)
  CPU: 24 × AMD Ryzen 9 5900X 12-Core Processor
  WORD_SIZE: 64
  LLVM: libLLVM-16.0.6 (ORCJIT, znver3)
Threads: 1 default, 0 interactive, 1 GC (on 24 virtual cores)
```

**Additional context**

Add any other context about the problem here.

matthew-kapp, 2025-03-04:
@ven-k Are you working on this? I believe quite a bit of the library needs to be updated to be (: name) as expected by @mtkmodel. I am quite happy to slowly make some of these changes.

SebastianM-C, 2025-03-04:
I think that the issue here is that the macro does not like positional arguments and forces keyword arguments.

matthew-kapp, 2025-03-05:
@SebastianM-C Is it as simple then as adding a semicolon at the start?

`function linear_interpolation(x1::Real, x2::Real, t1::Real, t2::Real, t)` -->
`function linear_interpolation(; x1::Real, x2::Real, t1::Real, t2::Real, t)`

It can't be this simple? Would this be a breaking change and therefore ill-advised?

JavierJJR, 2025-04-15:
Hi everyone,

Is there any update on this topic? I'm wondering if anyone has managed to resolve it or found a way to combine the @mtkmodel build to work around the issue.

rbborges, 2025-04-15:
> Hi everyone,
> 
> Is there any update on this topic? I'm wondering if anyone has managed to resolve it or found a way to combine the @mtkmodel build to work around the issue.

I haven't heard anything so far, and honestly I haven't looked at the latest documentation in case there's a different approach to the problem.

JavierJJR, 2025-04-15:
Wow, I was hoping it had either been solved or that a workaround had been found. I guess we’ll have to wait then :(

JavierJJR, 2025-04-15:
Thank you very much for your responses. I'm new to the Julia ecosystem, and honestly, everything is very new to me. I've been trying to figure this out all morning, but I haven't found a solution — everything I've tried has failed spectacularly.

matthew-kapp, 2025-04-29:
@SebastianM-C So is it possible to call a @component function with positional arguments inside a macro?

SebastianM-C, 2025-04-29:
No, the macro does not support positional arguments at all. I'm exploring some workarounds in https://github.com/SciML/ModelingToolkitStandardLibrary.jl/pull/388