SciML/ModelingToolkitStandardLibrary.jl #345: Automatically generate docs from component definitions

langestefan, 2024-11-25:
I've been going through the code / docs quite a bit these past few days, because I'm adding some components I need. 

**Is your feature request related to a problem? Please describe.**

To be blunt, the documentation is in a poor state. Some components have no documentation (`HeatingResistor` for example), some components have incomplete documentation and most documentation is of poor quality.

**Describe the solution you’d like**

Instead of manually putting docstrings at the top of component definitions I suggest we directly generate documentation from the fields inside `@mtkmodel`. In particular `description`, `bounds`, `guess`, `unit` and any other fields that are available from the metadata: [Symbolic Metadata](https://docs.sciml.ai/ModelingToolkit/dev/basics/Variable_metadata/#symbolic_metadata). This avoids the need to type everything twice. We can still have a docstring at the top.

ChrisRackauckas, 2024-12-03:
Yes, definitely agreed this would be a nice thing to add. That's a more of an MTK issue to improve the macro though.

langestefan, 2024-12-03:
Let me know if you have any ideas and where I can contribute.