SciML/ModelingToolkitStandardLibrary.jl #371: Dependency Symbolics from v6.29.2 to v6.30.0 is breaking

matthew-kapp, 2025-03-11:
**Describe the example**

- It seems the change of dependency Symbolics from v6.29.2 to v6.30.0 is breaking.
- New PR's after #368 have tests failing unrelated to the PR itself. Example #349 .
- If I pin Symbolics to v6.29.2 as done in #369 , the tests no longer fail.

**Summary of CI test results 👇**

Copied from "tests (Lts, Core):

- PR #368 : [0c5d862f] Symbolics v6.29.2 : ERROR: LoadError: Some tests did not pass: 572 passed, 2 failed, 0 errored, 2 broken.
- PR #349 : [0c5d862f] Symbolics v6.30.0 : ERROR: LoadError: Some tests did not pass: 507 passed, 2 failed, 6 errored, 0 broken.
- PR #369 : [0c5d862f] Symbolics v6.29.2: ERROR: LoadError: Some tests did not pass: 572 passed, 2 failed, 0 errored, 2 broken.

I have modified the Project.toml in PR #369 to pin Symbolics to v6.29.2, but this is obviously not a long-term fix

matthew-kapp, 2025-03-11:
@AayushSabharwal Could you perhaps have any insight into this?

AayushSabharwal, 2025-03-11:
Yeah, new functionality was introduced in Symbolics@6.30 via SymbolicUtils@3.17 that is buggy in ways that weren't caught by tests.

matthew-kapp, 2025-03-11:
> Yeah, new functionality was introduced in Symbolics@6.30 via SymbolicUtils@3.17 that is buggy in ways that weren't caught by tests.

The feat: add @cache macro?

AayushSabharwal, 2025-03-11:
Yep

closed 2025-03-14 by matthew-kapp